artemisamsx/artemisa #92: Overall design of GFX9920

apoloval, 2019-06-25:
From existing designs as the attached one, do a basic design to convert the component video output of TMS9928/29 to RGB and/or composite video. From it, identify the different models we will produce. Some examples:

- GFX9920A(RGB output)
- GFX9920B(CVBS output)
- GFX9920C(SCART output)

[TMS9929 RGB Adapter.zip](https://github.com/apoloval/artemisa/files/3324620/TMS9929.RGB.Adapter.zip)

apoloval, 2019-09-19:
Some notes about the investigation. 

I cannot interpret the proposed schematic. Things that are not clear to me:
- `Vref` is wired to `Pb` (tagged as `U`) on sync signals. Why?
- `Vref` is undetermined on non-sync signals. `U4D` is open during color transmission. And its output is connected to `U6`, the amplifier that generates `Vref`, without any pull up or down resistor. Thus, the input of `U6` is floating. 

I think it will be much better to try with [this other schematic from TI about how to connect TMS9928 to RGB monitors](http://spatula-city.org/~im14u2c/vdp-99xx/e3/SPPA004A_9928-29_9118-28_Interface_to_Monitors.pdf). We can adapt it to use a LM1881 in combination with 4066s to move the color levels to ground on color burst stage. Also, there are some fixes we should apply to the schematic:

- We have to consider the document says `R55` variadic resistor must be set to 4K. It's a typo. It means `R54`. 
- For green color generation, the resistor values are not the best. This configuration gives `G = 1.65Y - 0.34R - 0.31B`, far away from the theoretical equation of `G = 1.69Y - 0.5R - 0.18B`. Using `R44=15K` and `R45=12K` we obtain `G = 1.64Y - 0.5R - 0.21B`, much closer and better.

apoloval, 2019-09-19:
About schematic linked in description, I think I have some answers to my questions. 

`Vref` is not undetermined during non-sync signals. The capacitor attached to the non-inverting input of the amplifier is there for a reason. During sync signals, the capacitor is charged with the voltage coming from `Pb`. And `Vref` has that very value. When sync signal is over, `U4D` gets open, and the voltage in non-inverting input of U6 is the same we had during sync signal. Since the impedance of U6 inputs is extremely high, the capacitor is discharged very very slowly. So it can maintain the voltage level of `Pb` we had in sync signal during the color transmission phase. 

Thus, what the author expected from `Vref` is to maintain the value `Pb` has during sync for all the time. 

`U6` amplifier is required to have that. We cannot simply remove it and connect capacitor `C14` to `Vref`. If we would do so, the current demand of `Vref` in the whole circuit would discharge the capacitor quickly. And we would loose the voltage level. The amplifier decouples the impedance, producing a stable `Vref` signal.

apoloval, 2020-06-04:
I have been making some experiments with a MAX4031 and the design we have in c099530525ccdaa5f0d8a3df457993505f70c926. Which is basically the circuit attached in the description. And these are the results:

[image]

We are very close. What I did not is to remove the backporch sections from the Pb and Pr signals. 

During this period, I learned a lot about op-amps. Some important lessons learned:

- Most classic op-amps are designed to have wide power rails. From 12V to -12V, from 5V to -5V, etc. With 5V power supply, a single supply op-amp is required. It is easy to find modern parts that fit. But they use not to have DIP packages. 
- Most classic op-amps have a output voltage swing that is prevents the output to be close to the power rails. E.g, a ±12V op-amp use to support output voltages around ±9. Those op-amps that get close to the power rails are called rail-to-rail. Of course, if we use a single supply of 5V, we will need it. 
- Bandwidth. Most op-amps indicate this parameter, which usually means the maximum operating frequency for a gain of +1. The gain and the operating frequency are inversely proportional. Thus, if you need a gain of +2, the max operating frequency is divided by 2. Since MSX video has a pixel clock below 5Mhz, and sum amplifiers need a gain of +2, the op-amp must have a bandwidth above 10Mhz. 

Considering these requirements, LM318 is not an option. I used the MAX4031 with good results, but this is a SOIC-only part. The [AD8032](https://www.analog.com/media/en/technical-documentation/data-sheets/AD8031_8032.pdf) satisfies the requirements. The big problem is its 5.07€/unit. And we will need at least 2. 

I shared this data with the Retrocacharreo community. And this opened a discussion about having a single 5V power supply in the Artemisa. Many expansion hardware for sound generation and processing will not work because they depend on the ±12V power rail. If we consider this requirement, the design would be much simpler. Because LM318 will match the requirements powered by ±12V. Even though, we could use a power pump such as the [MAX870](https://www.maximintegrated.com/en/products/power/charge-pumps/MAX870.html)(available in DIP) to generate a -5V rail, what would make the LM318 suitable again.

apoloval, 2020-06-04:
I just opened a new issue to investigate the introduction of ±12V power rails: #150.

apoloval, 2020-06-04:
Other important considerations. I could understand better the DC restore phase described in the TMS document. 

The Y signal is clamped to the voltage level of the sync pulse using a clamping diode. The process is well explained [in the Wikipedia](https://en.wikipedia.org/wiki/Clamper_(electronics)#Positive_biased). The circuit before the anode is there to generate a bias voltage that can be adjusted with a potentiometer. 

The Pb and Pr signals cannot be clamped with this technique because it is not assured the lowest voltage corresponds to the sync periods. The paper suggests to use the sync signal separated from Y to clamp using the analog switches. If the analog switch connects GND to the capacitors only during these periods, this will clamp the input signals to the level they have during sync. 

After understanding all this, I think trying a different clamping approach is worth it. The clamping diode technique might be not necessary, because the LM1881 is actually giving us the sync pulse and we can use it to clamp with the analog switches are we do with Pb and Pr. 

I have to try that in the laboratory.

apoloval, 2020-06-04:
MAX870 is not available in DIP. I think it was a typo in the Maxim DB. The datasheet confirms it is not. 

[MAX660](https://datasheets.maximintegrated.com/en/ds/MAX660.pdf) is a similar part. But it is 4.5€/unit. For this price, it is better to keep using the AD8032 with a single 5V power supply.

apoloval, 2020-06-04:
I have been checking the RGB transformation detailed in the TI paper. 

The red component is totally okay. We have implemented it in the same way as the paper suggests. All fine. 

I have some doubts about the blue component. According to the paper, the VDP divides the B-Y signal by 2. This is the reason to put a resistor in B-Y half of the value of Y while summing for red color. That makes sense. As result, the op-amp would be generating R/3. That's what the paper says. And my calculations confirm it. Thus, the op-amp must be configured with a gain of +3. That's what the paper says. However, the text says the potentiometer must be set to 4K, which results in a gain of +2. This could be another typo in the paper. The point is that the bias resistor must be half of the feedback resistor. In our design, R19 must be 2K4 instead of 5K1. 

This is probably the reason because of blue is too dark. It is undersaturated.

apoloval, 2020-06-05:
Yes, totally. The dark blue is now much closer to the right color after fixing the value of R19 in 686e681ad0ffc10423c5a80336c54b447de093fe.

apoloval, 2020-06-08:
Important note. The LM1881 is not able to drive the CSYNC output and feed enough voltage for the CD4066 at the same time. 

Even using a large resistor at the TV CSYNC output, the voltage drops below the high level threshold of the 4066's control input. 

I had to use a NPN transistor in emitter-follower configuration as a buffer.

apoloval, 2020-07-15:
Okay. I have a working circuit using MAX4031. The colors are not exactly the ones I would expect, but they are fair enough. 

Well. MAX4031 is not valid for GFX9920 card because it is a SMD part. And one of the principles of 100 series is that it only uses TH parts. 

We could give a try to the LM318 if we get a 12V rail. But this is not available in 100 series. And I think it wouldn't be wise to wait for its introduction.

Another chance is to introduce a step up circuit in the very same graphics card. I tested the MC34063 circuit suggested in https://github.com/apoloval/artemisa/issues/150 and it works quite well. I have to extend the experiment to use the LM318 and see how it looks in the monitor.

apoloval, 2020-07-17:
Bad news. Only a single 12V rail is not enough. The voltage swing of the LM318 is huge. More or less 2-3V away from the power rails. This means a LM318 powered by 12V-0V only can handle signals above 2-3 volts and below 9-10 volts. 

Since the input signal is between 1 and 3 volts, it doesn't work. 

I am afraid we will need a voltage inverter to obtain the -12V to get it.

apoloval, 2024-11-17:
I've been having a look at this after a few years. 

[image]

Well, now that the plan is to have +12V/-12V rails and use the LM318 as [the original TI paper ](http://spatula-city.org/~im14u2c/vdp-99xx/e3/SPPA004A_9928-29_9118-28_Interface_to_Monitors.pdf) suggests, I think I should test their design first. The current circuit design is too cumbersome and it's likely overcomplicated. 

> Important note. The LM1881 is not able to drive the CSYNC output and feed enough voltage for the CD4066 at the same time.

That's because we are using the CSYNC signal to drive a 4066 to produce the Vref. This signal, however, is not used in the TI's original design. Again: let's see if we can make the TI approach work before.

apoloval, 2024-11-18:
Combining the colors with resistors doesn't sound to work very well. If you have a reference to the circuit, I'd like to have a look at it. 

I didn't see that implementation using the 9929. As far as I see, it's similar to the approach suggested by TI in their paper, but using BJT transistors instead of op-amps. 

[image]

I will have a study it carefully later. But as you said, this one uses the LM1881 to produce a Vref signal. And in my experiments, it drops a lot of current from it. So it have to be buffered. If we can get rid of the Vref signal generation, the better.

Cptscorcher, 2024-11-18:
This is the circuit I mentioned and outputs a composite signal. If I get a spare 5 minutes I'll try it out. But yeah not hopeful the quality would be any good.
[image]
[https://github.com/stefanberndtsson/tms9929-avr](url)

I don't have any LM1881's to try out the ElfCube circuit but looking at the image on their GitHub wiki the quality doesn't look too bad.

Cptscorcher, 2024-11-20:
Oh yes that is nice, I noticed that is what's used with the [https://tms-rgb.com/](url)  board.
I'm only a simple electronic tech so I don't fully understand the problem with the CYSNC - 4066 connection? As far as I can tell the LM1881 can supply the voltage/current for the 4066 which needs 1.5-3.5v and sucks up only 1uA the LM1881 can source/sink up to 1.6mA. Or am I missing something here?

apoloval, 2024-11-20:
> I'm only a simple electronic tech so I don't fully understand the problem with the CYSNC - 4066 connection? As far as I can tell the LM1881 can supply the voltage/current for the 4066 which needs 1.5-3.5v and sucks up only 1uA the LM1881 can source/sink up to 1.6mA. Or am I missing something here?

The problem is not the 4066, but the 1881. Apparently, the CSYNC signal that goes to the TV set drains too much current. So the 1881 cannot keep the expected voltage level of the 4066 inputs. And the interruptor doesn't work as expected. 

I know this is something I observed in my oscilloscope, but it was a few years ago and I cannot remember all the details. Maybe the TV set is not well connected. Or maybe those TV sets typically drain a lot of current. I've also seen in the [LMH1251 datasheet](https://www.ti.com/lit/ds/symlink/lmh1251.pdf) how it suggests to use buffers to the VGA connector, as the current is excessive for the IC to drive it without it. 

[image]
[image]

apoloval, 2024-11-20:
BTW, the TMS-RGB also have buffered outputs:

[image]

More precisely, the [THS7374
](https://www.ti.com/product/THS7374), a 4-Ch SDTV (Component and Composite) Video Amplifier with 9.5MHz Filters and 6dB Gain.

apoloval, 2024-11-20:
I am having a look at Mouser and LCSC, and the LM1881 is not available anymore in DIP encapsulation. Only SMD can be found. 

I think it's time to try [this circuit based exclusively on BJT transistors](https://www.msxarchive.nl/pub/msx/mirrors/hanso/hwdoityourself/rgbmsx1.pdf):

[image]

Cptscorcher, 2024-12-12:
Did you have a chance to try the transistor circuit?

apoloval, 2024-12-12:
Nope! Not yet. 

Do you bring yourself to try it?

Cptscorcher, 2024-12-17:
No not yet. Got a couple of weeks leave over Christmas coming up soon. This is the last thing to do and then I can see the Artemisa working. Planning to play a lot of Farming Simulator '25 during the break but still hope to have time :)

apoloval, 2026-11-05:
The breadboard tests with the MCP6294 were successful. There is a lot of noise, mostly caused by the breadboard itself and the Dupont wires, but the image is perfectly visible and the colors are right.

The schematic and the PCB have been updated accordingly, and the first revision of the board is now released as [GFX9920 Rev0](https://github.com/artemisamsx/gfx9920/releases/tag/rev0), including the Gerber files ready for fabrication.

About the LM1881: the original part is discontinued in DIP package, but HGSEMI manufactures a compatible LM1881N in DIP-8 that can be found at LCSC ([C518923](https://www.lcsc.com/product-detail/C518923.html)). That is the part to use for this board.

The last step before closing this issue is to build the Rev0 board and test it in a real Artemisa computer.