artemisamsx/artemisa #149: Suggestion to add license

Redlolz, 2020-05-27:
Am I allowed to use this project for my own designs, or is it under a open source license?

apoloval, 2020-07-15:
Sure! Thanks so much for the reminder. And apologies for taking so much time to reply. I was not watching my own repo, sorry!

I just added [the license file](https://github.com/apoloval/artemisa/blob/master/LICENSE.txt). This is the [TAPR Open Hardware License](https://en.wikipedia.org/wiki/Open_Hardware_License). In summary, you can use the designs (documentation) from this repository and the products (hardware) derived from them under the following conditions: 

- You may modify the documentation and make products based upon it.
- You may use products for any legal purpose without limitation.
- You may distribute unmodified documentation, but you must include the complete package as you received it.
- You may distribute products you make to third parties, if you either include the documentation on which the product is based, or make it available without charge for at least three years to anyone who requests it.
- You may distribute modified documentation or products based on it, if you:
  - License your modifications under the OHL.
  - Include those modifications, following the requirements stated below.
  - Attempt to send the modified documentation by email to any of the developers who have provided their email address. This is a good faith obligation—if the email fails, you need do nothing more and may go on with your distribution.
- If you create a design that you want to license under the OHL, you should:
  - Include the OHL document in a file named LICENSE.TXT (or LICENSE.PDF) that is included in the documentation package.
  - If the file format allows, include a notice like "Licensed under the TAPR Open Hardware License (www.tapr.org/OHL)" in each documentation file. While not required, you should also include this notice on printed circuit board artwork and the product itself; if space is limited the notice can be shortened or abbreviated.
  - Include a copyright notice in each file and on printed circuit board artwork.
  - If you wish to be notified of modifications that others may make, include your email address in a file named "CONTRIB.TXT" or something similar.
- Any time the OHL requires you to make documentation available to others, you must include all the materials you received from the upstream licensors. In addition, if you have modified the documentation:
  - You must identify the modifications in a text file (preferably named "CHANGES.TXT") that you include with the documentation. 
  - That file must also include a statement like "These modifications are licensed under the TAPR Open Hardware License."
  - You must include any new files you created, including any manufacturing files (such as Gerber files) you create in the course of making products.
  - You must include both "before" and "after" versions of all files you modified.
  - You may include files in proprietary formats, but you must also include open format versions (such as Gerber, ASCII, Postscript, or PDF) if your tools can create them.

Enjoy it!

closed 2020-07-15 by apoloval

Redlolz, 2020-07-19:
Thank you very much!