artemisamsx/artemisa #150: Introduce ±12V power supply rails

apoloval, 2020-06-04:
I am afraid ±12V is not as expendable as I initially supposed. We had a fruitful discussion in the Retrocacharreo channel in Telegram about the topic. And many hardware for sound and video processing demands these rails to feed their op-amps. 

The goal of this issue is to investigate a solution for this problem and apply it in 101 and 201 motherboard series.

apoloval, 2020-06-04:
Here we have [a really nice video about the topic](https://www.youtube.com/watch?v=giGRrODKJSE). 

Some buck converters are available at low price. Such as the [MC34063A](https://docs.rs-online.com/e64e/0900766b8138583a.pdf). However, I still have to figure out how it works.

apoloval, 2020-06-04:
The [TI datasheet](http://www.ti.com/lit/ds/symlink/mc34063a.pdf?ts=1591264916555) provides a better description of the step up and down circuits. It looks like we could use a pair of these to obtain +12V and -12V from a single source of +5V. 

I've found [this article explaining how to convert +5V to -12V](https://www.multisim.com/content/BCyDrk9tGaJe5NAcySJvaP/inverter-5v-to-12v-mc34063/). And here we have [an schematic to step up from +5V to +12V](https://easyeda.com/modules/5V-to-12V-MC34063_3bfd29b7b410498fb2b7099be5e41eaa). With this, we could get the 12V rails we need.

apoloval, 2020-06-04:
In fact, the TI datasheet actually provides an example to get -12V from 5V:

[image]

msx-solis, 2020-06-12:
[image]

msx-solis, 2020-06-12:
Hi, I fount this through internet, I hope it helps you.
aditionally there's a circuit based on an LT1054 what can supply +12v -12v from Vcc (+5v). its on the following link:
[https://retrobrewcomputers.org/n8vem-pbwiki-archive/0/35845334/48860720/35845587/Printing%20MSXCartReader-sch.pdf](url)

apoloval, 2020-06-13:
Wow! Many thanks for the references! Especially for the LT1054. 

The [datasheet](https://www.ti.com/lit/ds/symlink/lt1054.pdf?ts=1592065465468&ref_url=https%253A%252F%252Fwww.google.com%252F) shows an example to generate +12V/-12V from +5V. Even thou I don't understand it quite well (I cannot identify the outputs in the diagram):

[image]

The circuit from the MSX Cart Reader linked about is this one:

[image]

I think it is based on this one from the datasheet:

[image]

However, the TI datasheet says this is a doubler with a loss around 1V. So this would be +9V/-9V if I am not wrong.

apoloval, 2020-06-13:
Oh, I think there is a huge copy&shit in the TI datasheet. The figure 23 is exacty the same as the figure 26:

[image]

It is like they copied the wrong diagram and the converter we are looking for is missing. 

Fortunately, there are other vendors who have the right circuit [in their datasheet, as Analog Devices](https://www.analog.com/media/en/technical-documentation/data-sheets/LT1054-1054L.pdf):

[image]

And yes, it is exactly the one shown in the MSX Cart Reader schematic. 

Both parts are similar. But I have the feeling that the LT1054 requires less external components than the MC34063. Or at least they are more standard ones (no inductors required for the 1054!).

apoloval, 2020-06-13:
Pricing:
- LT1054: 2.99€/unit in Mouser. 
- MC34063A: 0.54€/unit in Mouser.  

The difference is huge. As we need 2 units, the 1054 is 6x as expensive than the 34063.

apoloval, 2022-11-02:
Recent updates. A completely different approach could be using a external power supply. The problem is to find a triple output one with 5V, +12V and -12V outputs. The problem is that I couldn't find one like this. Until now. 

We have the [Mean Well RT-50B][1]. A enclosed power supply with 5V, +12V and -12V outputs. This could be embed into the computer case (if fits) such that the computer is directly powered by an AC cable. Or, it could be put into a separated case and connected to the computer case using a DIN cable. This unit is around 30€ in AliExpress. 

In fact, Mean Well also has external power supplies with the same outputs that uses DIN cables. That's the case of [GP25A13D-R1B][2]. It is more expensive, [around 50€ including taxes in Digikey][3]. 

I think this is an excellent solution, as it is a sort of "out of design". We will not have to care about it anymore, assuming the motherboard receives power inputs of 5V, +12V and -12V. Period. 

We have to think about how to provide a triple power switch.... Or not have any, as the original 101 design. 

[1]: https://www.meanwell.com/Upload/PDF/RT-50/RT-50-SPEC.PDF
[2]: https://www.meanwell.com/webapp/product/search.aspx?prod=GP25A
[3]: https://www.digikey.es/es/products/detail/mean-well-usa-inc/GP25A13D-R1B/7703337

apoloval, 2022-11-02:
At least we know they exist: 

[image]

msx-solis, 2022-11-02:
I think you shown a DPDT switch but you need a 3PST (if exists) or a 3PDT like this:
https://es.rs-online.com/web/p/interruptores-de-palanca/2740429

apoloval, 2022-11-02:
> I think you shown a DPDT switch but you need a 3PST (if exists) or a 3PDT like this: https://es.rs-online.com/web/p/interruptores-de-palanca/2740429

Yep, 3PST is an odd animal. But some vendors make 4P2T switches that could do the job, as [this one](http://www.xiejia.net/ProductAll.asp?MID=210).

apoloval, 2022-11-02:
Some interesting parts:
- [ALPS SSSF040800](https://lcsc.com/product-detail/Slide-Switches_ALPSALPINE-SSSF040800_C388871.html)
- [ALPS SPUN19C400](https://lcsc.com/product-detail/Push-Switches-Horizontal-Type_ALPSALPINE-SPUN19C400_C402824.html)
- [E-Switch PBH6UEENAGX](https://lcsc.com/product-detail/Push-Switches-Horizontal-Type_E-Switch-PBH6UEENAGX_C273425.html) (too expensive perhaps)
- [G-Switch PS-42F43](https://lcsc.com/product-detail/Push-Switches-Horizontal-Type_G-Switch-PS-42F43_C2848940.html)
- [G-Switch PBS-42D02](https://lcsc.com/product-detail/Push-Switches-Horizontal-Type_G-Switch-PBS-42D02_C2848956.html)

Obtained from [LCSC website](https://lcsc.com/products/Push-Switches-Horizontal-Type_11275.html).

apoloval, 2022-11-21:
Another interesting finding. Mean Well also has products such as [MDD02L-12][1]. This is a DC-DC unregulated converter encapsulated in a module with a PCB surface of 19.5x9.8mm. 5V inputs, producing a DUAL output of ±12V with a maximum current of ±83mA (MSX standard requires ±50mA for ±12V rail of all slots combined). 

It costs <7€ in Mouser. Just requires a few standard capacitors and one inductance to work:

[image]

It is designed to isolate the input and output circuits, but after Googling about this subject I am pretty sure we can connect `Vin-` and `Com` together to have a common ground for input and output circuits. The benefit of isolation will be neutralized doing it, but we need a common ground due to MSX design restrictions. 

[1]: https://www.meanwell.com/webapp/product/search.aspx?prod=MDD02

apoloval, 2022-11-21:
The [DPUN02L-12][1] is even better. Industrial grade instead of medical, but smaller and cheaper (5.07€ in Mouser). 

[1]: https://www.meanwell.com/webapp/product/search.aspx?prod=DPUN02

apoloval, 2026-07-10:
For now, this is superseded by #154.