artemisamsx/artemisa #154: Custom power supply

apoloval, 2022-06-08:
In model 101, a single 5V DC input power source was used. The system depends on a external voltage source, typically provided by a brick adapter. This solution presents several drawbacks:

- Brick adapters are a bullshit. It is really hard to find a vendor whose adapters do not generate a source out from 4.75-5.25 volts range. Quite often, they go down to 4.65 volts. 
- No ±12V power line. This is something already reported in #150. 

Perhaps the way to go is to provide a custom power supply, with its 220/115 VAC to 30 VAC transformer, rectifier bridge, and voltage regulators using 7812, 7912 and 7805 parts.

apoloval, 2022-06-08:
I investigated a little bit. And there are some interesting AC transformers in Mouser such as [this][1] and [this][2]. 

However, they are a little bit expensive. Around 15€. 

I also checked if Chinese parts could do the job. They are around 8€, but I could not find anything PCB-mounted. 

There is also another alternative: design a 5V adapter and complement it with a ±12V step-up circuit as suggested in #150. But that is probably extremely power-inefficient. 

[1]: https://www.mouser.es/datasheet/2/177/160_161-1389957.pdf
[2]: https://www.mouser.es/datasheet/2/177/162_164-1390053.pdf

apoloval, 2022-06-08:
Another interesting finding. Hi-Link produces [a encapsulated power module that converts 220/115 VAC to 5 VDC][1]. This is cool. But there is no solution for ±12V rails. 

[1]: https://www.hlktech.com/en/Goods-16.html

apoloval, 2022-06-08:
Also, the classic AC to ±12V dual voltage regulator circuit is [explained here][1]. 

[1]: https://www.eleccircuit.com/power-regulator-12v_-12v-by-7812-7912/

apoloval, 2022-06-09:
Oh, this is an interesting discovery. The [Block VC 10/2/15][1]  is a encapsulated block device that can convert 220 VAC into a dual output of 15 VAC. There are many encapsulated block devices that does this, but they are typically low power. But this one has a rate of 10 VA. It is also unexpensive: [8€ in RS Online][2]. However, it is quite large. 51x43x34mm. Perhaps we can find something smaller... 

Bingo! Having a look to the [Block catalog of PCB transformers][3] we can find the [VC 5/2/15][4]. Rated as 5 VA, with 44x33x37mm dimensions. [7€ in RS Online][6]. The [PT 4.5/2/15][5] is also a good fit. 4.5 VA, short-circuit proof and 37x44x33mm. There is also a VB family, but the max rate is 3.2 VA. Probably not enough. 

Reading related stuff, I discovered the power rate refers to parallel output connections. This is, if you use it as a single 15 VAC output. If serial output connection is used to have ±15 VAC, the power rate for the positive rails (the one that will feed 7805 and 5 VDC rail) is half. This means we likely need a 10 VAC device. But we have to double check how these power translates into the circuit after voltage regulators. Perhaps we are aiming too high and we don't really need such large power rates. 

[1]: https://www.block.eu/en_EN/productversion/vc-10215/
[2]: https://es.rs-online.com/web/p/transformadores-de-pcb/3472622
[3]: https://www.block.eu/en_EN/products/transformers/pcb-transformers/
[4]: https://www.block.eu/en_EN/productversion/vc-50215/
[5]: https://www.block.eu/en_EN/productversion/pt-45215/
[6]: https://es.rs-online.com/web/p/transformadores-de-pcb/3472559

apoloval, 2022-06-09:
Side note. A AC/AC transformer with a rectifier bridge backed up by a capacitor to smooth the output, produces a unregulated DC source whose value matches the peak voltage of the AC transformer. 

This means for a 12Vac transformer: $12V * \sqrt{2} = 16.97V$. Most 7812 datasheets indicate they can operate with input voltages above 14V. Thus, a  ±12 VAC transformer would do the job.

In case of Block transformers, this means using [VC 5/2/12][1] or [PT 13/2/12][2]. They have the same size as 15V equivalents. But we might expect to be cheaper, and for sure we will have a wiser power usage. 

[1]: https://www.block.eu/en_EN/productversion/vc-10212/
[2]: https://www.block.eu/en_EN/productversion/pt-13212/

apoloval, 2022-06-09:
Side note to the side note. The rectifier bridge is made by diodes that have a voltage drop. We can assume from 0.7 to 1.0 per diode, having 2 diodes with forward current at the same time. That's from 1.4V to 2.0V drop. So peak-to-peak output voltage from the transformer must be $>16V$. 

Close to the limit. The fact that the signal is smoothed by a large capacitor also means a relative voltage drop. More calculations are needed.

apoloval, 2022-06-09:
Interesting reference. [TI Application report on Input and Output Capacitor Selection][1]. Interesting to calculate the properties of the capacitors used in a voltage regulation circuit. 

[1]: https://www.ti.com/lit/an/slta055/slta055.pdf

apoloval, 2022-06-09:
Another interesting reference. A [Stack Exchange answer explaining how power rating affects an AC/DC regulator using 7805][1]. 

[1]: https://electronics.stackexchange.com/a/328202/215644

apoloval, 2022-06-09:
An interesting finding. I tried to simulate the AC transformer and wave rectifier in Falstad simulator:

[image]

- Source has peak voltage of 17V, 50Hz. 
- Filter capacitor is 10mF. 

If we force the situation and put a load that sinks 1A of current:

[image]

The voltage at the capacitor falls to 13.97V. That's close the the limit of the 7812. At that rate, it will have problems to deliver 12V at its output. 1 ampere seems to be a hard limit. 

The bad news come from the ripple current hitting the capacitor. As they reach almost 12A. That's huge!

[image]

I think we have to calibrate the power consumption better to find realistic limits. 

The Falstad circuit:

[rectifier.circuitjs.txt](https://github.com/artemisamsx/artemisa/files/8868676/rectifier.circuitjs.txt)

apoloval, 2022-06-09:
Having a look to schematics of old MSX systems to see how they do it. 

Philips VG8020 uses an AC transformer with a dual coils: 

[image]

One is used for low-power, ±12V rails. The other for high-power 5V rail. 

The Spectravideo SVI-728 does a similar thing. The transformer has a dual output of 9Vac and 16Vac:

[image]

This is how they essentially get reasonable power losses due to voltage drops. We should take this way also, I am afraid.

apoloval, 2022-06-09:
[Block VB 10/2/15][1] could be used for ±12V rails. It is just 32x27mm. The power rate of 1VA is enough for the expected 30mA. 

[Block PT 7.5/1/8][2] could be used for 5V rail. It is 51x43mm. Rated at 7.5VA, despite we need 8VA for 8V operation. 

[VC 10/1/8][3] has the same dimentions. Rated at 10VA, exceeds the required 8VA. 

In any case, we will need a lot of room in the PCB for all this. 

[1]: https://www.block.eu/en_EN/productversion/vb-10215/
[2]: https://www.block.eu/en_EN/productversion/pt-7518/
[3]: https://www.block.eu/en_EN/productversion/vc-1018/

apoloval, 2022-06-09:
Of course, there is still the option of using a bump-up regulator such as the [MAX743][1]. It can use a 5V input to produce a ±12V rail. However, it is quite expensive. [15€ per unit][2].

[1]: https://www.maximintegrated.com/en/products/power/switching-regulators/MAX743.html
[2]: https://www.mouser.es/ProductDetail/Maxim-Integrated/MAX743EPE%2

apoloval, 2022-06-10:
Some useful maths. 

If we use a transformer to 8Vac for 5V rail regulation: 

$V_{pk} = 8V \cdot \sqrt{2} = 11.3V$

The voltage drop is defined as:

$V_{drop} = \frac{I \cdot T_{cycle}}{C}$

We know the voltage cannot go below 7V, which is the minimum input voltage for the 7805 to work:

$V_{drop} = 11.3V - 7V = 4.3V$

Thus, considering a max current of 1A and 50Hz of frequency (period of 10ms) our filter capacitor should be at least:

$C > \frac{I \cdot T_{cycle}}{V_{drop}} > \frac{1A \cdot 10ms}{4.3V} > 2.32 mF$

apoloval, 2022-06-13:
Good news. The power connector + power switch is the [Schurter CMF1][1]. And it is [cheaper in RS Online: 6€ if you pick 10 units][2]. 

[1]: https://www.schurter.com/en/datasheet/CMF1__CMF4
[2]: https://es.rs-online.com/web/p/conectores-iec/4982782

apoloval, 2022-06-13:
CMF1.1111.12 is the second best choice. With isolation cover, but mounting from above. This means the box should have a mounting hole exactly where the connector goes. This is good, because it will be more robust. But that is a harder constraint. 

| Vendor | Price 1ud | Price 10uds |
| --- | --- | --- |
| Mouser | 10.37€ | 9.93€ |
| Farnell | 0.98€ | 8.69€ |
| RS Online | 7.11€ | 5.45€ |
| Digikey | 8.06€ | 8.06€ |

apoloval, 2022-06-15:
Maths for the 12V rail. 

If we use a transformer to 15Vac for ±12V rail regulation: 

$V_{pk} = 15V \cdot \sqrt{2} = 21.2V$

We have to consider a voltage drop from the rectifier bridge about 1.4V, so:

$V = 21.2V - 1.4V = 19.8V$

The voltage drop is defined as:

$V_{drop} = \frac{I \cdot T_{cycle}}{C}$

We know the voltage cannot go below 14.5V, which is the minimum input voltage for the 7812 to work:

$V_{drop} = 19.8V - 14.5V = 5.3V$

Thus, considering a max current of 100mA and 50Hz of frequency (period of 10ms) our filter capacitor should be at least:

$C > \frac{I \cdot T_{cycle}}{V_{drop}} > \frac{100mA \cdot 10ms}{5.3V} > 18.8 uF$

The energy dissipated by the regulator is:

$P = I \cdot V_{drop} = 100mA \cdot 5.3V = 0.53W$

apoloval, 2022-06-15:
A good candidate for bridge rectifiers is the [[DF005M-E3/45][1]. Rated at 1A, 50V, for only half an Euro. 

[1]: https://www.mouser.es/datasheet/2/427/dfm-1768184.pdf

apoloval, 2022-06-15:
Bad news. Typical rectifier diodes have a drop voltage of 1.1V. Including the [DF005M-E3/45][1] and the [1N4007][2]. 

It means the calculations made above are wrong. We have to fix them, and likely use other output voltages in the transformers. At least for 8Vac. 

[1]: https://www.mouser.es/datasheet/2/427/dfm-1768184.pdf
[2]: https://www.mouser.es/datasheet/2/849/1n4001-2577232.pdf

apoloval, 2022-06-15:
Repeating maths for 5VC generation. 

If we use a transformer to 8Vac for 5V rail regulation: 

$V_{pk} = 8V \cdot \sqrt{2} = 11.3V$

We have to consider a voltage drop from the rectifier bridge about 2.2V, so:

$V = 11.3V - 2.2V = 9.1V$

The voltage drop is defined as:

$V_{drop} = \frac{I \cdot T_{cycle}}{C}$

We know the voltage cannot go below 7V, which is the minimum input voltage for the 7805 to work:

$V_{drop} = 9.1V - 7V = 2.1V$

Thus, considering a max current of 1A and 50Hz of frequency (period of 10ms) our filter capacitor should be at least:

$C > \frac{I \cdot T_{cycle}}{V_{drop}} > \frac{1A \cdot 10ms}{2.1V} > 4.76 mF$

The maximum energy dissipated by the regulator is:

$P = I \cdot V = 1A \cdot (9.1V - 5V) = 4.1W$

apoloval, 2022-06-15:
Repeating maths for the 12V rail. 

If we use a transformer to 15Vac for ±12V rail regulation: 

$V_{pk} = 15V \cdot \sqrt{2} = 21.2V$

We have to consider a voltage drop from the rectifier bridge about 1.4V, so:

$V = 21.2V - 2.1V = 19.1V$

The voltage drop is defined as:

$V_{drop} = \frac{I \cdot T_{cycle}}{C}$

We know the voltage cannot go below 14.5V, which is the minimum input voltage for the 7812 to work:

$V_{drop} = 19.1V - 14.5V = 4.6V$

Thus, considering a max current of 100mA and 50Hz of frequency (period of 10ms) our filter capacitor should be at least:

$C > \frac{I \cdot T_{cycle}}{V_{drop}} > \frac{100mA \cdot 10ms}{4.6V} > 21.7 uF$

The maximum energy dissipated by the regulator is:

$P = I \cdot V = 100mA \cdot (19.1V - 12V) = 0.71W$

apoloval, 2022-06-15:
We have a candidate for fuse holder: [Schurter 0031.8201][1]. 0.86€ in Mouser. 

[1]: https://www.mouser.es/datasheet/2/358/typ_OGN-14523.pdf

apoloval, 2022-06-16:
MORE bad news. I am afraid that the size of the power transformers is huge for the target box and PCB dimensions.

[image]

[image]

I still don't know if the free space after putting the transformers will be enough or not.

apoloval, 2022-06-16:
The [RAC10-12DK/277][1] is a PCB-mounted, AC/DC converter with a 85-305Vac input and dual output of ±12Vd at 420mA. The size is 52.5x27.4x23.0mm. [15€ in RS Online][2]. 

Quite expensive. At this price range, we can get a [MAX743][3]. 

[1]: https://docs.rs-online.com/2f24/0900766b8161b79a.pdf
[2]: https://es.rs-online.com/web/p/fuentes-de-alimentacion-conmutadas/1655094
[3]: https://www.maximintegrated.com/en/products/power/switching-regulators/MAX743.html

apoloval, 2022-06-17:
I've been Googling about MC34063, and it reveals it is a very popular chip for DC/DC conversion. There are tons of tutorials. It is so popular that there are tons of calculators to find the values for the auxiliary components, [like this one][1]. 

I've been also considering that having PCB-mounted AC/DC transformation is overkill. We actually have wall power adapters that can produce 5Vdc outputs. Moving it our from the board as we did in model 101 sounds wiser. 

However, if we look at specs, we see that wall power adapters are bullshit. They have typically a total output regulation of ±5%, with ripple voltages up to 300mVp-p. This means the actual output voltage can vary from 4.75V to 5.25V. This is something that matches the observations we had while selecting wall power adapters for model 101. And also the experience: in model 101, any touch to the power connector causes noise in the power line that translates into image flickering on the VDP.

In contrast, the [LDE05-20B05][2] performs much better. Line regulation of ±0.5%, load regulation of ±2%. Overall voltage accuracy of ±2%, with voltage ripple of 50mVp-p. This means the output voltage should be strictly in the 4.9-5.1Vdc range. 

Using a wall power adapter is cheaper. Not only because we can obtain them from 6€ (compared to the 8€ of the LDE), but also that we don't need the 6€ of the switched power connector. However, its poor performance makes me think it is wiser to go with the LDE regulator unless we find a way of compensating the poor output regulation of the brick power adapters. 

[1]: http://gmsystems.com/switching-reg-calculator-for-mc-34063-or-mc33063.html
[2]: https://docs.rs-online.com/6ace/0900766b816b4d25.pdf

apoloval, 2022-06-17:
Excellent finding! Another approach could be to use a regulated (or not) 9Vdc wall adapter, and use switching DC/DC converter to generate a stable signal. 

The [K7805M-1000R3][1] has 8-36Vdc inputs and 5Vdc output at 1A. ±0.2% of line regulation, ±0.4% of load regulation, ±1.5% of voltage accuracy and a voltage ripple of 25mVp-p. [3€ in RS Online][2]. The problem is that they have no stock. 

Other part: the [VR10S05][3]. Same input/output specs, except ±2% of voltage accuracy and voltage ripple of 75mVp-p. Also [3€ in RS Online][4]. 

[1]: https://docs.rs-online.com/dc16/A700000006631882.pdf
[2]: https://es.rs-online.com/web/p/reguladores-de-conmutacion/1934004
[3]: https://docs.rs-online.com/7616/A700000007304872.pdf
[4]: https://es.rs-online.com/web/p/reguladores-de-conmutacion/2124689

apoloval, 2022-06-17:
RS Online also vendors [K7805-2000R3L][1]

[1]: https://docs.rs-online.com/fc3c/A700000006631878.pdf. Same specs, but output rated at 2A, ±0.4% of line regulation, ±0.5% of load regulation and voltage ripple of 30mVp-p. [5€ in RS Online][2]. 
[2]: https://es.rs-online.com/web/p/reguladores-de-conmutacion/1934043

apoloval, 2022-06-17:
In fact, there is a K7XXX version for 12V: [K7812][1]. This is much simpler than using the [MC34063A][3] to pump-up the voltage from 5V to 12V. As it requires almost no external component. And the price is similar: [2.5€ in RS Online][2]. 

[1]: https://docs.rs-online.com/994b/A700000006631906.pdf
[2]: https://es.rs-online.com/web/p/reguladores-de-conmutacion/1933980
[3]: https://docs.rs-online.com/e64e/0900766b8138583a.pdf

apoloval, 2022-06-17:
In fact... what the heck? The problem on using linear regulators is that, at the current rate we will have in the 5V rail, we will dissipate too much energy in form of heat. But thanks to using a switching regulator such as K7805, that's not a problem at all. The ±12Vdc rails will consume below 100mA. We might even use linear regulators with 15Vdc input voltage, and they will not consume more than 300mW. We will likely not even need heatskinks!

apoloval, 2022-06-17:
According to this heat sink calculator, with 15Vdc input no heat sink is needed. Not even with 24Vdc input. 

[1]: https://www.petervis.com/electronics/Voltage_Regulator_Heatsink/heat-sink-calculator.html

apoloval, 2022-06-17:
I know I said this before, but I think we have a winner.

| Part | Price |
| ---- | ----- |
| Wall AC/DC power adapter from 15V to 24V | [8€ in RS Online][1] |
| K7805 or compatible (VR10S05 is pin compatible, and external components are the same) to regulate to 5V | 3€ |
| A K7812 (or L7812) to regulate to +12V | 3€ (or 0.5€) |
| A L7912 to regulate to -12V | 0.5€ |
| Rest of well known elements: barrel connector, LED, power switch... | |
| Total | 14.5€ (or 12€ if use L7912) |

[1]: https://es.rs-online.com/web/p/adaptadores-ac-dc/1753306

apoloval, 2022-06-17:
Wait a second. Wake up! L7912 cannot be used unless we have a negative voltage rail. It cannot invert the voltage 😢 

The -12V rail cannot be achieved by that. I think we will have to use the damned [MC34063A](https://docs.rs-online.com/e64e/0900766b8138583a.pdf).

apoloval, 2022-06-17:
Another interesting finding. [MAX765][1] is a DC/DC inverter that convers 3-16Vdc input into fixed -12Vdc output. With just a few capacitors, one **small** inductance and one diode:

[image]

[1]: https://docs.rs-online.com/b429/0900766b81253d09.pdf

apoloval, 2022-06-17:
Also [MAX636][1], which is even simpler. 

[image]

[1]: https://datasheets.maximintegrated.com/en/ds/MAX635-MAX637.pdf

apoloval, 2022-06-17:
MAX parts are expensive after all. MAX765 is 5€, and MAX636 is not even in catalog. 

Another good finding: [TC7662][1]. It inverts the input voltage for just 2€ with really minimum external components:

[image]

It has a relatively high output impedance, what causes a voltage drop when current increases. However, not a big deal in the range of 50mA as we expect from it. 

[image]

[1]: https://docs.rs-online.com/3444/0900766b813bf648.pdf

apoloval, 2024-11-17:
Come back to think in this problem, I've made a (yet another) research.

In summary: switching converters are efficient, small and noisy. Linear converters are inefficient, large and silent. The challenge consists in finding a good tradeoff. We know that efficiency is good as it means low heat dissipation. But noise is bad for analog devices, and we must take into account that an MSX computer generates video and sound outputs. 

As far as I can [read in this TI paper](https://www.ti.com/lit/ta/sszt239/sszt239.pdf), most designers typically use a switching regulator followed by Low Drop-Off (LDO) linear converter to have a low-noise power rail. That's a very interesting approach. Also, some vendors provide low-ripple DC-DC switching regulators. What have to be taken into account as well. 

In summary, I think the right way to go is having a brick AC/DC converter with a crappy regulation. And then use it to have a low-ripple and high-quality 5V/+12V/-12V rails. If low-ripple switching regulators are enough, go for it. If they are too noisy, use LDOs. 

Some interesting products:
- [WPME-FDSM](https://www.we-online.com/components/products/datasheet/173011235.pdf) is a switching regulator, Vout 5V for Vin >8V. Ripple is 17.5mVpp. 
- [Meanwell DPRN01](https://www.meanwell-web.com/en-gb/meanwell-sprn01-dprn01/) is a switching regulator, dual outputs of +12V/-12V. Ripple is 75mVpp. 
- [TI TPMS8291x](https://www.ti.com/product/TPSM82913) is a low-noise and low-ripple buck module. Can be used to generate 5V from an input of >3V and a ripple of micro-volts. TI claims it is a replacement for the switching + LDO pipeline in a single encapsulation. It's not TH, but it can be used for other designs, as the noise is extremely low.

apoloval, 2024-11-17:
Having a huge offer of LDOs in the market, I think the perfect scenario would be having switching regulators to obtain 6V, +13V and -13V, and then low-ripple LDOs to convert them into 5V, +12V and -12V with almost zero ripple. It's just a matter of finding the right parts in TH encapsulation.

Cptscorcher, 2024-11-18:
As far as I know the -/+ 12v was only used for the RS232. Perhaps instead of using up valuable PCB real estate with fancy switching regulators. We could either develop an onboard RS232 and there is precedence for this with the YIS-503  [https://www.msx.org/wiki/Yamaha_YIS-503IIR](url) which has the firmware in ROM at 0x4000 - 0x5fff. Or another option could be a new cartridge with a MAX232 IC. 
Either onboard or cartridge I think it would make the RS232 more accessible than trying to find an OEM RS232 cartridge like the Sony HBI-232. IMHO :)

apoloval, 2024-11-18:
BTW, Meanwell DPRN01 is not suitable as it is an isolated converter. So we cannot mix the grounds with other converters.

apoloval, 2024-11-18:
There are some interesting non-isolated buck converters for outputs of 5V and 1A. Just listing a few:
- [Traco TSR 1-2450E](https://www.mouser.es/datasheet/2/687/tsr1e_datasheet-1842504.pdf). 3.64€, Vin > 7V, ripple 80mVpp typ. 
- [Recon R-78E5.0-1.0](https://www.mouser.es/datasheet/2/468/R_78E_1_0-1711201.pdf). 3.74€, Vin > 8V, ripple 120mVpp max. 
- [Wurth 173010578](https://www.we-online.com/components/products/datasheet/173010578.pdf). 6.78€, Vin > 8V, ripple 25mVpp typ. 
- [Recon R-785.0-1.0](https://www.mouser.es/datasheet/2/468/R_78_1_0-1709942.pdf). 7.79€, Vin > 6.5V, ripple 20mVpp typ, 30mVpp max. 

However, for -12V the options are very limited. Meanwell offers [the N7812-2](https://www.mouser.es/datasheet/2/260/MeanWell_08192024_N78_2_spec-3483609.pdf), but it is not real dual output. It just supports negative configuration, so you need 2 units at 5.87€ each.  The remaining parts converting to -12V starts at 13.35€. 

I am thinking the best choice is to use a power module for 5V rail and have +12V/-12V rails using low-ripple switching regulators with external inductance and diodes.

apoloval, 2024-11-18:
About using switching regulators with external components: large inductances are not cheap. They start at 3.49€ in Mouser.

apoloval, 2024-11-19:
I am afraid that, at least for the +-12V rails, a custom converter circuit must be designed using a switching voltage regulator and maybe LDOs at the end.

Mouser offers switching voltage regulators starting at 0.57€ with [OnSemi MC34063](https://www.mouser.es/datasheet/2/308/1/MC34063A_D-2315484.pdf). However, the most popular part is the [LM2596](https://www.ti.com/lit/ds/symlink/lm2596.pdf). It is used in cheap Chinese voltage regulator boards on sale in AliExpress and friends. It became so popular that many different vendors implement it, including first brands as OnSemi but also many Chinese manufacturers from $0.42 at LCSC. 

The 2596 is a buck converter, but it admits inverter topologies. So it can be used to generate a -12V rail, or a -13V connected to a LDO. It could be used also for the 5V rail, but considering how cheap and quiet are the power modules listead above, I don't think it worth. 

For the 12V rail, we can use the 2596 as well as long as the input voltage is above, likely around 15V. However, it means the drop-off for the 5V rail is 10V. What makes the conversion more inefficient and noisy. Perhaps it's better to use a boost converter in such a case. The [MC34063](https://www.onsemi.com/download/data-sheet/pdf/mc34063a-d.pdf) seems to be a very popular option. And in fact, it also allows buck conversion. And it's incredible cheap! The implementation from STMicroelectronics from 0.25€ each at LCSC, 0.57€ at Mouser. The limit is 1.5A compared to the 3A of 2596. But it's enough for the Artemisa.

As a curiosity, STMicro's 34063 datasheet suggest a circuit to have a +-12V rails. Unfortunately, it requires a transformer coil:

[image]

apoloval, 2024-11-19:
Interesting finding. The [Wurth 173010578](https://www.we-online.com/components/products/datasheet/173010578.pdf) is designed to allow negative voltage generation:

[image]

Just taking into account that the input voltage range changes:
- For -12V, Vin must be in range 9-20V. 
- For 12V, Vin must be in range 16-36V.
- For 5V, Vin must be in range 8-36V. 

If we use the same part for all voltage rails, the input voltage should be in range 16-20V. 

I also discovered that Meanwell also have a similar part: the N78 series. Ripple=100mVpp from 4.04€. It can be used for negative volt regulation using the same approach of changing the pins. The input voltage range is the same: 16-20V. 

[image]

apoloval, 2024-11-19:
Another unexpensive approach is using a [7662 charge pump](https://www.mouser.es/datasheet/2/268/21469a-3442046.pdf) to invert a 15V input into a -15V line. And then use linear regulators to produce +-15V. TC7662 can be found from 2.49€ in Mouser. 

The charge pumps have a significant output impedance, so they output less than desired voltage. But even if they produce -14V for a 15V input, it is enough for a LDO to produce -12V.