artemisamsx/artemisa #155: Any Progress

Cptscorcher, 2024-08-22:
Not really an issue but just wondering if there are any further updates for this project? Looks like there has been a lot of work put into this project and I'm keen to build one of these but just not sure if either the 101 or 201 boards are mature enough to be working.

apoloval, 2024-08-26:
To be honest, the project is currently stalled from the product perspective. 

The development of the 101 board is finished. Both the design and the implementation are tested and the computer works like a charm. I think that's the model to go if you want to build a homebrew computer by your own. In the other hand, the model 201 is under development. There is nothing usable there yet. 

When I say model 101 is stalled from the product perspective I mean that the necessary steps to provide it as a DIY product you can purchase from the website are not completed yet. The pending parts are the documentation. My plans consists on having an extensive handbook explaining not only how the computer can be assembled but also the fundamentals of every single circuit in the computer. A sort of dissertation of 1980's computer design that IMO is 99% of the product value. 

However, if you are interested in building the computer by your own and you have the knowledge and skills, feel free to generate the Gerber files of the boards and build it by yourself. Every few months I make some contributions to the documentation. But I cannot tell when it will be completed and DIY kits ready to be shipped. Same for the design of the 201 model.

Cptscorcher, 2024-08-27:
Thanks apoloval. I have done some mods on the 201 version and hope to fork and upload it at some point. But for now it's probably best to go with something tested so have ordered 5 boards. 
The documentation is great, nice to see something more than a one line readme.md. I'm an electronic tech by trade and work as a technical writer. I'll send you an e-mail with more detail but I am keen to contribute to writing the docs. I could start with the build instructions when I finally get all the bits.

closed 2024-11-17 by apoloval